ohmtech-rdi/eurorack-blocks #455: Misaligned header pins / daisy patch library footprint / kivu12.kicad_pcb

da6353, 2022-07-24:
Header pins in library footprint (DaisyPatchSM) for daisy patch are slightly misaligned, see attached screen shots for comparison.

[image]
[image]
[image]

ohmtech-rdi, 2022-07-25:
Hi @da6353 !

If you refer to the plated through-holes zig-zag pattern, that actually comes from the Electro-smith Eagle layout, and was just converted this way in Kicad.

Electro-smith did that actually on purpose: If you use headers with squared pins, this will slightly tighten the header in place when hand-soldering. Having used that we find it pretty smart! And so the power connector is not done this way, so we are thinking of making the same thing for it.

@da6353 Do you have any trouble with that zigzag pattern when soldering?

Please keep us posted, thanks!

da6353, 2022-07-26:
Thanks!

I make a generic matching 12HP front user panel protoboard, and also a 14HP
version ('kivu14'), which has pin headers realigned, so that standard pcb
protoboard can be used, for the front user panel.

Thanks, Don

[image: image.png]

[image: image.png]

ohmtech-rdi, 2022-07-26:
Hi @da6353 

> I make a generic matching 12HP front user panel protoboard, and also a 14HP
version ('kivu14'), which has pin headers realigned, so that standard pcb
protoboard can be used, for the front user panel.

Ah I see! But the pin headers that have this zigzag pattern are meant for the Daisy Patch Submodule, and are going to the back of the `kivu12` board, right.

Having a protoboard for the front user panel is indeed a smart idea!

As for names, the idea was to use lakes in Africa, suffixed with the HP size.

Thanks!

closed 2023-01-08 by ohmtech-rdi