ohmtech-rdi/eurorack-blocks #459: Eurorack-blocks for Max doesn't build properly on maxOS Monterey "bash: erbb: command not found"

ohmtech-rdi, 2022-08-07:


ohmtech-rdi, 2022-08-07:
On macOS Monterey, `zsh` is used by default. But Max still uses `bash`.

We need to source our environment with the `PATH` to erbb. So not only we have to fill `.zshrc` but also `.bash_profile` in the Max case.

ricardomatias, 2022-12-20:
Do you have any objections in having [fish](https://fishshell.com/) support?

ohmtech-rdi, 2022-12-20:
Hi @ricardomatias !

Thanks for the suggestion. I'm not sure exactly on how this could address this issue regarding Max. Could you please elaborate on that?

If this is a general suggestion, I'll be happy to make a new issue for you on this topic.

Please keep us posted. Thanks!

ricardomatias, 2022-12-26:
It's at most tangentially related, so more of a general suggestion :) Best to open a new issue.

ohmtech-rdi, 2022-12-26:
Please don't hesitate to open it yourself, as you can probably put more details than me, or otherwise tell me to do it for you. Thanks!

10T25, 2023-01-03:
Type this command in a bash shell:
chsh -s /bin/zsh
It will switch your shell from bash to zsh, then no more error from max environment. Everything seems to build correctly.