ohmtech-rdi/eurorack-blocks #460: Rack segfaults when attempting to load a plugin generated via faust

jazamatronic, 2022-10-25:
G'day,

Rack is instantly crashing when trying to load a plugin that was built by erbb from faust sources - such as the flanger example.

I have fixed it locally [here](https://github.com/ohmtech-rdi/eurorack-blocks/compare/main...jazamatronic:eurorack-blocks:rack_segfault?expand=1#diff-3fea5fd94ce514fc3d31e7eb873a06e9d46e053cee88f8cefcd4534b8810d2c8) by initializing dsp memory during the init.

I can create a pull request if you'd like?

Cheers

ohmtech-rdi, 2022-10-26:
Hi @jazamatronic !

Thanks a lot for your patch. I'm not sure how it could possibly work before!
I'll create the Pull request.

Thanks!

ohmtech-rdi, 2022-10-26:
Ah so apparently, it depends on the Faust version. The Faust version doesn't generate the `memoryCreate` function for at least version 2.20.2, which is the current version `apt`:

```
Get:16 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libfaust2 amd64 2.20.2~ds0-1 [1838 kB]
Get:17 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 faust-common all 2.20.2~ds0-1 [29.4 MB]
Get:18 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 faust amd64 2.20.2~ds0-1 [4071 kB]
```

The version I have locally is:

```
$ faust --version
FAUST Version 2.41.1
```

And this one actually generates the `memoryCreate` function.

Not sure what to do, a bit surprised a breaking change was introduced, but I don't know all the mechanics behind.

@sletz Could you please help us with that? Thanks!

jazamatronic, 2022-10-26:
G'day @ohmtech-rdi,

Thanks for taking a look.  I'm running a very new clone/build of faust, running on Linux:

```
$ faust --version
FAUST Version 2.50.6
```

Cheers

ohmtech-rdi, 2022-10-26:
Thanks for letting me know your version, indeed that's very new, but I can already reproduce the problem with a stock `faust` build using `brew` on macOS 10.15 so that's going to be problematic as well for every new users.

I guess there might be a way to keep backward compatibility, will check with @sletz

ohmtech-rdi, 2022-10-26:
- Closed in favor of #464 (It's not possible to rewrite history for gh PRs, and I needed to correct style on first commit)

PR #464 (2022-12-06): Fix faust memory create
Fix missing internal buffers allocation by calling `memoryCreate` before the DSP instance is inited, following the [documentation](https://faustdoc.grame.fr/manual/architectures/#defining-and-using-a-custom-memory-manager) on the new `-mem` option.

The `memoryCreate` function is a new function, but which must be called, in order for the DSP to correctly instantiate.

To keep compatibility with pre-2.41, we are checking whether the function exists or not at build time and call it if available.

- Fixes #460
closed 2022-12-06 by ohmtech-rdi