ohmtech-rdi/eurorack-blocks #461: Pots and CV_IN are not connected to the right adc pins in daisy_micropatch

jazamatronic, 2022-10-25:
G'day,

I tried building the flanger faust example and programming it to a patch.init(), but wasn't getting the expected behaviour.
In the end it came down to the adc pin definitions for the Pots and CV_IN jacks being swapped around.
I have a fix in my forked branch [here](https://github.com/ohmtech-rdi/eurorack-blocks/compare/main...jazamatronic:eurorack-blocks:patch_init_pot_fix?expand=1#diff-c7ece60b0df9fe48338356ad5de7b7c9720e628ecc213b173934d3ea8256e75a)
I can create a pull request if desired?

Cheers

ohmtech-rdi, 2022-10-26:
Great, thanks again for reporting.

I'll need to get my hand on my patch.init() to verify all that.

I think I'll take it from there. Thanks!

jazamatronic, 2022-10-26:
Ok.  I built my own from the electro-smith board files, maybe they're not up to date or something.  I have a spare board and faceplate I could post if you felt like getting your soldering iron out...

ohmtech-rdi, 2022-10-26:
That's very nice of you. But don't worry, I have one of those boards too. It's one of the pre-release ones I believe though, so I'll need to check with Electrosmith if they actually changed some routing on it (but I don't think so).

I just need to find it in my cartons 🙂 (just moved to a new office)