ohmtech-rdi/eurorack-blocks #469: Issue when using more than 4 Pots in gen~

10T25, 2022-12-27:
I have a project in gen~ that uses 7 Pots, 5 Trims. The first four Pots are OK, but not the others, neither the Trims. they simply don't respond. No solder join, every connection checked.

ohmtech-rdi, 2022-12-29:
After some investigations:

To reproduce, `kivu12` compatible program must loaded from the `QSPI` section.
In that case, the `A8` pin on the Daisy Patch SM is stuck.

Electro-smith has already a fix for that:
- https://github.com/electro-smith/libDaisy/pull/548

I'll probably make a branch to test out their new bootloader, but that requires some work as we are still on `libDaisy` v4.0.0 and we are aiming for at least `v5.3.0` so we might have some breaking changes to look into.

ohmtech-rdi, 2023-01-04:
So the branch is sitting on `wip-bootloader-v6`, and there was no big breaking changes. Only one struct now has a default constructor, so we can't use designated initializers anymore.

The v6 bootloader actually seems to completely solve that issue.

Now waiting until it gets released.