ohmtech-rdi/eurorack-blocks #474: Problem with Configure

omar-karray, 2023-02-15:
in the setup phase of Eurorack-blocks 
I'm on Mac 13.0.1 (22A400) Apple M1 Pro, 
i have installed python `brew install python` 
then `erbb setup` on the cloned folder all the steps right :)
where I execute `erbb configure` on the `drop` folder  I get this problem. 

```
Traceback (most recent call last):
  File "/Users/omarkarray/Documents/EurorackDev/eurorack-blocks/build-system/scripts/erbb", line 466, in <module>
    sys.exit (main ())
              ^^^^^^^
  File "/Users/omarkarray/Documents/EurorackDev/eurorack-blocks/build-system/scripts/erbb", line 421, in main
    import erbui
  File "/Users/omarkarray/Documents/EurorackDev/eurorack-blocks/build-system/erbui/__init__.py", line 15, in <module>
    from .generators.vcvrack.panel import Panel as vcvrackPanel
  File "/Users/omarkarray/Documents/EurorackDev/eurorack-blocks/build-system/erbui/generators/vcvrack/panel.py", line 15, in <module>
    from ..detail.panel import Panel as detailPanel
  File "/Users/omarkarray/Documents/EurorackDev/eurorack-blocks/build-system/erbui/generators/detail/panel.py", line 24, in <module>
    import cairocffi
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cairocffi/__init__.py", line 48, in <module>
    cairo = dlopen(
            ^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/cairocffi/__init__.py", line 45, in dlopen
    raise OSError(error_message)  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: no library called "cairo-2" was found
no library called "cairo" was found
no library called "libcairo-2" was found
cannot load library 'libcairo.so.2': dlopen(libcairo.so.2, 0x0002): tried: 'libcairo.so.2' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibcairo.so.2' (no such file), '/usr/lib/libcairo.so.2' (no such file, not in dyld cache), 'libcairo.so.2' (no such file), '/usr/lib/libcairo.so.2' (no such file, not in dyld cache)
cannot load library 'libcairo.2.dylib': dlopen(libcairo.2.dylib, 0x0002): tried: 'libcairo.2.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibcairo.2.dylib' (no such file), '/usr/lib/libcairo.2.dylib' (no such file, not in dyld cache), 'libcairo.2.dylib' (no such file), '/usr/lib/libcairo.2.dylib' (no such file, not in dyld cache)
cannot load library 'libcairo-2.dll': dlopen(libcairo-2.dll, 0x0002): tried: 'libcairo-2.dll' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibcairo-2.dll' (no such file), '/usr/lib/libcairo-2.dll' (no such file, not in dyld cache), 'libcairo-2.dll' (no such file), '/usr/lib/libcairo-2.dll' (no such file, not in dyld cache)
```

If someone could help will be good !

omar-karray, 2023-02-15:
Found the Issue Dependency conflict between the install phyton3.11 on Mac and the installed one by brew.

closed 2023-02-15 by omar-karray

ohmtech-rdi, 2023-02-15:
Hi @omar-karray yes it's what it looks like.

The python installed by homebrew is configured to read into `/opt/homebrew/lib` to find libraries, and here your installation of python is a different one, so it can't find the `libcairo.2.dylib`library.

Please try the following as a workaround:
```
DYLD_LIBRARY_PATH=/opt/homebrew/lib erbb configure
```
Does it work?

ohmtech-rdi, 2023-02-15:
Ah sorry, I've read to quickly.

Could you please tell how you made to select the right version of python? I think some other people would be glad to know.

omar-karray, 2023-02-16:
hey @ohmtech-rdi 
Thank you for your reactivity. the process is easy 
remove all the python library 
`sudo rm -rf /Library/Frameworks/Python.framework/Versions/3.11`

then remove the app from the application folder 

`sudo rm -rf “/Applications/Python 3.11"`

then break that simlink that make the hole trouble :) 

`ls -l /usr/local/bin | grep ‘../Library/Frameworks/Python.framework/Versions/3.11’`

after that make sure that python3 is installed via home-brew 

`brew info python3`  
if not installed 

`brew install python3`

and you are good to go. 
finally will it be more interesting to have it dockerized with a docker image for erbb ? 
all the best 
Omar