ohmtech-rdi/eurorack-blocks #476: erbb build simulator fails to build on arch linux

nckwntzl, 2023-03-02:
It appears to be having trouble finding `rack.hpp` however I checked and the file does exist in `~/eurorack-blocks/submodules/vcv-rack-sdk/include/`

```
make: Entering directory '/home/lethargic/eurorack-blocks/samples/drop/artifacts/simulator'
ACTION UI
ACTION VCV Rack
CXX src/vcvrack/BoardGeneric.cpp
CXX plugin_vcvrack.cpp
CXX Drop.cpp
In file included from ../../../../include/erb/vcvrack/BoardGeneric.h:308,
                 from ../../../../src/vcvrack/BoardGeneric.cpp:12:
../../../../include/erb/vcvrack/BoardGeneric.hpp:32:10: fatal error: rack.hpp: No such file or directory
   32 | #include <rack.hpp>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:170: Release/home/lethargic/eurorack-blocks/src/vcvrack/BoardGeneric.cpp.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ../../../../include/erb/vcvrack/BoardGeneric.h:308,
                 from ../../../../boards/kivu12/simulator/BoardKivu12.h:16,
                 from ../../../../boards/kivu12/BoardKivu12.h:20,
                 from ../../artifacts/DropUi.h:21,
                 from ../../Drop.h:13,
                 from ../../Drop.cpp:12:
../../../../include/erb/vcvrack/BoardGeneric.hpp:32:10: fatal error: rack.hpp: No such file or directory
   32 | #include <rack.hpp>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:184: Release/home/lethargic/eurorack-blocks/samples/drop/Drop.cpp.o] Error 1
In file included from ../../../../include/erb/vcvrack/BoardGeneric.h:308,
                 from ../../../../boards/kivu12/simulator/BoardKivu12.h:16,
                 from ../../../../boards/kivu12/BoardKivu12.h:20,
                 from ../../artifacts/DropUi.h:21,
                 from ../../Drop.h:13,
                 from ../plugin_vcvrack.cpp:16:
../../../../include/erb/vcvrack/BoardGeneric.hpp:32:10: fatal error: rack.hpp: No such file or directory
   32 | #include <rack.hpp>
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:177: Release/home/lethargic/eurorack-blocks/samples/drop/artifacts/plugin_vcvrack.cpp.o] Error 1
make: Leaving directory '/home/lethargic/eurorack-blocks/samples/drop/artifacts/simulator'
Traceback (most recent call last):
  File "/home/lethargic/eurorack-blocks/build-system/scripts/erbb", line 466, in <module>
    sys.exit (main ())
  File "/home/lethargic/eurorack-blocks/build-system/scripts/erbb", line 430, in main
    build ()
  File "/home/lethargic/eurorack-blocks/build-system/scripts/erbb", line 275, in build
    erbb.build_simulator_make_target (module, cwd, configuration)
  File "/home/lethargic/eurorack-blocks/build-system/erbb/__init__.py", line 499, in build_simulator_make_target
    subprocess.check_call (cmd)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '--jobs', '--directory=/home/lethargic/eurorack-blocks/samples/drop/artifacts/simulator', 'install']' returned non-zero exit status 2.
```

ohmtech-rdi, 2023-03-02:
Hi @nckwntzl !

Sorry you encountered an error, and thanks for the console output for the error.

I don't have Arch Linux installed so I'll most probably need assistance to solve that problem.
Here I would supposed this has something to do with `make`.

Could you please run `make -v` and tell me what it returns? Thanks!

nckwntzl, 2023-03-02:
> Hi @nckwntzl !
>
> Sorry you encountered an error, and thanks for the console output for the error.
>
> I don't have Arch Linux installed so I'll most probably need assistance to solve that problem. Here [...]

here is the output of `make -v`

```
GNU Make 4.4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

ohmtech-rdi, 2023-03-02:
Ok, that's a fairly recent version of GNU make. By any change do you have experience with Makefiles?

ohmtech-rdi, 2023-03-02:
Maybe could you please try:
```
$ cd ~/eurorack-blocks/samples/drop/artifacts/simulator
$ make -d install
```

Then please copy the output here, maybe first as a text file you can put as an attachment to your post.

nckwntzl, 2023-03-02:
I attached the output to the original post, as well as the makefile.

ohmtech-rdi, 2023-03-02:
That's great, thanks so much.

So in the Makefile:
```
RACK_DIR ?= ../../../../submodules/vcv-rack-sdk
...
FLAGS += -I$(RACK_DIR)/include
...
```

And in the output of the `make` command:
```
g++ -MMD -MP -std=gnu++2a -Wsuggest-override -fPIC -I../../../../include -I/usr/share/vcvrack/include -I/usr/share/vcvrack/dep/include -DARCH_LIN -fno-gnu-unique -O3 -DNDEBUG=1 -DRELEASE=1 -march=nocona -funsafe-math-optimizations -Wall -Wextra -Wpedantic -Werror -Derb_TARGET_VCV_RACK -Derb_BUFFER_SIZE=48 -Derb_SAMPLE_RATE=48014 -I../.. -I../../.. -c -o Release/home/lethargic/eurorack-blocks/src/vcvrack/BoardGeneric.cpp.o ../../../../src/vcvrack/BoardGeneric.cpp
```

`-I/usr/share/vcvrack/include` seems to indicate that you have a `RACK_DIR` in your environment variables.
Can you give it a try and try:
```
echo $RACK_DIR
```
I would expect this outputs:
```
/usr/share/vcvrack
```
Is that the case?

In that case could you please inspects the content of `/usr/share/vcvrack/include` and check if `rack.hpp` is there? I would guess not, but maybe you could have a file around to tell you which version of vcvrack you have installed there? Thanks!

ohmtech-rdi, 2023-03-02:
Ok, so provided you are using this [Pacman package](https://aur.archlinux.org/packages/vcvrack),
it has this [`profile.sh`](https://aur.archlinux.org/cgit/aur.git/tree/profile.sh?h=vcvrack) file that get installed [here](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=vcvrack#n139) so at your system level.

I'm not sure what to think about it, since the official VCV Rack repo is counting on `include` to [be there](https://github.com/VCVRack/Rack/blob/v2/plugin.mk#L18).

So that seems a bit a package mistake to me. I'll investigate a bit more.

ohmtech-rdi, 2023-03-02:
One work around would be to override `RACK_DIR`, for example please try the following from the `drop` sample folder:

```
RACK_DIR=~/eurorack-blocks/submodules/vcv-rack-sdk erbb build
```

nckwntzl, 2023-03-02:
There are 3 packages in the AUR, I tested two and one of them seems to only partially populate `/usr/share/vcvrack` while the other does not have that directory at all. I am now trying the package that builds from git, I will see where that takes me. 

> Ok, so provided you are using this [Pacman package](https://aur.archlinux.org/packages/vcvrack), it has this [`profile.sh`](https://aur.archlinux.org/cgit/aur.git/tree/profile.sh?h=vcvrack) file [...]

agreed that it is an issue with the packages, especially considering the different behavior between them.

> One workaround would be to override `RACK_DIR`, for example please try the following from the `drop` sample folder:
> 
> ```
> RACK_DIR=~/eurorack-blocks/submodules/vcv-rack-sdk erbb build simulator
> ```
> 
> Does it work?

I will test this after testing the build from git.

nckwntzl, 2023-03-02:
Overriding `RACK_DIR` did solve one problem, however I am still having errors with the build. Attached is the output
[output.txt](https://github.com/ohmtech-rdi/eurorack-blocks/files/10876359/output.txt)

FYI I am on gcc 12.2.1

nckwntzl, 2023-03-02:
using the following

> One workaround would be to override `RACK_DIR`, for example please try the following from the `drop` sample folder:
> 
> ```
> RACK_DIR=~/eurorack-blocks/submodules/vcv-rack-sdk erbb build simulator
> ```
> 
> Does it work?

as well as adding `#include <cstddef>` to `include/erb/detail/DoubleBuffer.h` got the build to work on gcc 12.2.1 under arch linux. I am closing this issue since everything was due to my build environment.

closed 2023-03-02 by nckwntzl

ohmtech-rdi, 2023-03-02:
Ok great, thanks. About `size_t`, it's because we should use `std::size_t` and I guess gcc 12.2.1 is now enforcing it.
I'll make a separate issue for that, because this is not because of your build environment, but our code is not fully standard compliant.

Thanks again for providing the outputs so we could solve this problem.

ohmtech-rdi, 2023-03-04:
@nckwntzl FYI:

- #478 

This exhibited the same `gcc` errors, so I could correct them and ensure CI now checks for them.