ohmtech-rdi/eurorack-blocks #482: Issue with Leds L8 to L16

10T25, 2023-04-11:
I'm experiencing an issue with BiLeds above L8.
I've written a very simple max patch and an erbui file to reproduce the issue with a kivu12 board.
The pot allows the BiLed to turn linearly from green to red (FCCW to FCW). No issue with L1 to L8.
With BiLed from L9 to L16 the behavior is different. Leds turn from solid orange (FCCW to 9) to green to red (9 to 3) then solid orange (3 to FCW).

[image]
[image]

ohmtech-rdi, 2023-04-11:
Hi @10T25 

Could you please give it a try with `Led` instead of `LedBi`? I think that would make it easier to see what is going on.

I think it's been a long time since I tested from L9, so there might be an issue, but I don't really see where.

Could you please also measure your resistors on L9 to L16 to ensure they are all around 1KΩ?

Thanks!

ohmtech-rdi, 2023-04-12:
Great, thanks for the test. I'm going to make a "kivu12 eval board" to be able to debug that. I should have done this a long time ago I think. In the meantime, I'll try to reroute the signal from another front panel.

ohmtech-rdi, 2023-04-16:
This is a work-in-progress. I've put everything that kivu12 supports. I still need to had the normalisation prove, and the PCB will be also the front panel to ease access to signals as well as keeping the price low. This will help me to debug this issue.

[image]

10T25, 2023-04-16:
I had this idea of exposing every in/out/pots .... some kind of super patch.init. You were faster than me!

ohmtech-rdi, 2023-04-17:
Got a few things to fix but that should be it:

[image]

10T25, 2023-04-18:
And with the normalisation probe! ;-) can't wait!!!

10T25, 2023-04-25:
Tested and validated. Bug fixed, well done.

PR #483 (2023-04-25): Fix led 16b value range to avoid hole between 4036 and 4094
This PR fixes what seems to be a "bug" on the PCA9685 12-bit led driver, where there is a "hole" beetwen the values from 4036 to 4094 (inclusive).

We had a previous fix for that (#473 ) but the bug was still there for pins from LED8 to LED15.

We fix that by scaling from 0 to 4035 (inclusive) instead of 0 to 4095. 4095 is anyway not full off in our case, so lacking the dynamic range from this "hole" doesn't seem to have a really important visual effect.

- Fixes #482