ohmtech-rdi/eurorack-blocks #556: Issue when setting up the environment on Windows

OmsInSerial, 2023-06-11:
Installation from scratch on a Windows 11 machine. Issue when executing the erbb setup command :

```shell
MINGW64 ~/eurorack-blocks (main)
$ python --version
Python 3.11.4

MINGW64 ~/eurorack-blocks (main)
$ pip --version
pip 23.1.2 from C:\Users\xxxxx\AppData\Local\Programs\Python\Python311\Lib\site-packages\pip (python 3.11)

``` 

```shell
MINGW64 ~/eurorack-blocks (main)
$ erbb setup
ERROR: Can not combine '--user' and '--target'
Extracting msys2_mingw64.zip...
Extracting kicad_minimal_windows_6.0.11-0.tar.gz...
Extracting gcc-arm-none-eabi-10.3-2021.10-win32.zip...
Installing python packages...
Traceback (most recent call last):
  File "C:\Users\xxxxx\eurorack-blocks\build-system\scripts\erbb", line 474, in <module>
    sys.exit (main ())
              ^^^^^^^
  File "C:\Users\xxxxx\eurorack-blocks\build-system\scripts\erbb", line 425, in main
    setup ()
  File "C:\Users\xxxxx\eurorack-blocks\build-system\scripts\erbb", line 188, in setup
    setup.install_python_requirements ()
  File "C:\Users\xxxxx\eurorack-blocks\build-system\setup\__init__.py", line 352, in install_python_requirements
    subprocess.check_call (
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\subprocess.py", line 413, in check_call
    raise alledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\xxxxx\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\\python.exe', '-m', 'pip', 'install', '--quiet', '--quiet', '--quiet', '--disable-pip-version-check', '--log', 'C:\\Users\\xxxxx\\eurorack-blocks\\build-system\\toolchain\\pip-install-pip.log.txt', '--target', 'C:\\Users\\xxxxx\\eurorack-blocks\\build-system\\toolchain\\python3-packages', 'pip==23.1.2']' returned non-zero exit status 1.

```

ohmtech-rdi, 2023-06-12:
Thanks for the report. That sounds to be Python from the Windows store.

Did you install Python as in the doc? If not please **do not install** python as said in the documentation, let's keep this configuration to see if we can support it.

ohmtech-rdi, 2023-06-12:
Also please tell me the output of the following:

(note the 3)
```
$ python3 --version

$ which python3

$ python3 -m pip --version
```

Thanks!

OmsInSerial, 2023-06-12:
So to be accurate : 
- Python installed as said in the doc (not through Windows store) is not  linked in GitBash. Tried this with no success : 

```shell
alias python='winpty python.exe'
``` 
 - When you launch Python in GitBash it launches Windows Store to install Python. Then Python is available with GitBash but with the issue described above.

So here is the output for the commands:

```shell
$ python3 --version
Python 3.11.4

$ which python3
/c/Users/xxxxx/AppData/Local/Microsoft/WindowsApps/python3

$ python3 -m pip --version
pip 23.1.2 from C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1264.0_x64__qbz5n2kfra8p0\Lib\site-packages\pip (python 3.11)

```  
Thanks

ohmtech-rdi, 2023-06-12:
Hi, @OmsInSerial could you please run the following commands and tell me what they return?

```
python3 -m pip config list

python3 -m pip config --global list

python3 -m pip config --user list

python3 -m pip config --site list
```

The [error message](https://github.com/pypa/pip/commit/716c9202eeecbf4601b39a8ab7bc7085c36e3883) is fairly old, so I suspect that Windows might set up some configuration for `pip`. I guess something like:
 
 ```
 [install]
 user = true
 ```
 Or something.
 
 Thanks!

OmsInSerial, 2023-06-13:
Added --no-user option to subprocess.check_call in '_init_.py'

```python
def install_python_requirements ():
   os.makedirs (PATH_PY3_PACKAGES)

   print ('Installing python packages...')

   # Start by installing a version of pip after 22.2. This version introduces
   # the 'universal2' that some our dependencies, like freetype-py, depends
   # on, in order to install the wheel we need.

   subprocess.check_call (
      [
         sys.executable,
         '-m', 'pip', 'install',
         '--quiet', '--quiet', '--quiet',
         '--disable-pip-version-check',
         '--log', os.path.join (PATH_TOOLCHAIN, 'pip-install-pip.log.txt'),
         '--target', PATH_PY3_PACKAGES,
         '--no-user',
         'pip==23.1.2',
      ]
   )

   # pip install --target does not support well shared namespaces. Here with
   # the bin/ folder which is used for `pip` binaries, which we don't use
   # so remove that folder before installing requirements.

   shutil.rmtree (os.path.join (PATH_PY3_PACKAGES, 'bin'))

   subprocess.check_call (
      [
         sys.executable,
         '-m', 'pip', 'install',
         '--quiet', '--quiet', '--quiet',
         '--disable-pip-version-check',
         '--log', os.path.join (PATH_TOOLCHAIN, 'pip-install-reqs.log.txt'),
         '--target', PATH_PY3_PACKAGES,
         '--no-user',
         '--report', os.path.join (PATH_TOOLCHAIN, 'pip-report.json'),
         '--requirement', os.path.join (PATH_ROOT, 'requirements.txt'),
      ],
      cwd=PATH_PY3_PACKAGES
   )
``` 

Seems to be OK now!
```shell
xxxxx@xxxxxx MINGW64 ~/eurorack-blocks (main)
$ erbb setup
     -------------------------------------- 790.5/790.5 kB 8.4 MB/s eta 0:00:00
     ---------------------------------------- 90.6/90.6 kB 5.4 MB/s eta 0:00:00
     ---------------------------------------- 1.9/1.9 MB 15.5 MB/s eta 0:00:00
Extracting msys2_mingw64.zip...
Extracting kicad_minimal_windows_6.0.11-0.tar.gz...
Extracting gcc-arm-none-eabi-10.3-2021.10-win32.zip...
Installing python packages...
Optimizing erbb...
Checking toolchain...
OK.

``` 

Also build the Drop sample whithout any prob to validate the environment.

ohmtech-rdi, 2023-06-13:
> Also, I had to manually set Errb and Eurorack-blocks packages paths in Max

Did you have an error if you didn't do that? If so, what was it?

ohmtech-rdi, 2023-06-13:
> Added --no-user option to subprocess.check_call in _init_.py

I've made this PR #558 yesterday for this, I guess the change is OK to merge anyway.

OmsInSerial, 2023-06-13:
OK thanks

closed 2023-06-13 by OmsInSerial