ohmtech-rdi/eurorack-blocks #597: windows / max "modulenotfounderror" "no module name soundfile found"

garrett-laroy-johnson, 2023-07-16:
Having this issue both when generating a VCV by saving a Max folder or when I go to init a new module in CL. 

[image]

[image]

ohmtech-rdi, 2023-07-16:
Hi @garrett-laroy-johnson

Did you run `erbb setup --with-max-support`?

In all cases please run it, and please copy/paste the output of this command here. In particular I'm interested in any errors, or if `erbb` could check the toolchain (because it tries to load soundfile at that time).

Thanks!

garrett-laroy-johnson, 2023-07-18:
Hi Rapheal,

I cloned a clean repo, reinstalled Python, and don't seem to be having that
issue any longer :)

ohmtech-rdi, 2023-07-19:
Hi @garrett-laroy-johnson Ok great, I'll closed the issue as magically solved 🙂

closed 2023-07-19 by ohmtech-rdi