ohmtech-rdi/eurorack-blocks #599: cairocffi failuire in macos build instructions

olilarkin, 2023-07-20:
hey!

when following the tutorial for macos with xcode. I got an error

```bash

~/Dev/eurorack-blocks$ erbb setup --with-xcode-support                ✹main
#·Packages·required·to·user·erbb·and·erbui
Extracting kicad_minimal_macos_6.0.11-0.tar.gz...
Extracting gcc-arm-none-eabi-10.3-2021.10-mac.tar.bz2...
Extracting toolchain_big_sur.tar.gz...
Adding Xcode support...
Installing python packages...
  Preparing metadata (setup.py) ... done
  Building wheel for cairocffi (setup.py) ... error
  ERROR: Failed building wheel for cairocffi
  ERROR: Failed cleaning build dir for cairocffi
ERROR: Could not build wheels for cairocffi, which is required to install pyproject.toml-based projects

```

in order to fix it, i removed the version number from cairocffi here:

https://github.com/ohmtech-rdi/eurorack-blocks/blob/341e72b52465b0cdf94ff4fb8a463d0c5c70fe94/requirements/build-system.txt#L5

```
-cairocffi>=1.4.0,<1.5.0
+cairocffi
```

ohmtech-rdi, 2023-07-21:
Hi @olilarkin 

Thanks for the report! I remember I added this `<1.5.0` because the build was failing on Windows, but I didn't investigate more on it, and I totally should as this is quite restrictive.

I think that was is happening is that a `1.4.x` wheel is not available for your platform, which might be a combination of machine architecture, macOS version as well as `python3` version, so please could you tell me:

- Which Apple computer model are you using?
- Which version of macOS are you running?
- Which `python3 --version` are you running?
- Which `python3` is running by running the command below:
```
python3 -c 'import os.path; import sys; print(os.path.realpath(sys.executable))'
```

Thanks!

olicarter, 2023-08-28:
I was also getting the same error.

- MacBook Air M1 2020 16GB
- Ventura 13.2.1
- Python 3.9.6
- `/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/bin/python3.9`

I noticed from getting this info that the last instance of Python existed within Xcode, so I manually deleted `/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/` and reinstalled Python from [https://www.python.org/downloads/macos](https://www.python.org/downloads/macos).

After doing that it built successfully 🎉 

The docs currently say to ensure that `which python3` doesn't include Xcode anywhere in the path, though mine didn't include Xcode but still failed. Would it be worth including the `python3 -c 'import os.path; import sys; print(os.path.realpath(sys.executable))'` check in the docs too?

ohmtech-rdi, 2023-08-28:
Hi @olicarter 

Great! Thanks for the feedback.
Yeah indeed, I guess that `which python3` returns `/usr/bin/python3`? Which I guess then itself is a symlink to the Python in your Xcode installation?

I guess I could indeed put this quickly in a troubleshooting section in the documentation, but I guess I could also do this in the [environment check priori to installation](https://github.com/ohmtech-rdi/eurorack-blocks/blob/main/build-system/setup/__init__.py#L80). What do you think?