ohmtech-rdi/eurorack-blocks #619: Wrong plugin path on windows (plugins-win-x64)

MathiasSchneider-fr, 2023-09-07:
Hello, I recently installed version 2.4.1 of VCV rack on Windows (it was a first installation, not an update, VCV rack had never been installed on this machine) and the installation worked for me. created a rack2/plugins-win-x64 directory and no rack2/plugins directory. So this plugins-win-x64 became the default directory in which VCV rack searches for plugins.
The construction of the Eurorack Blocks examples subsequently created the rack2/plugins directory (since it did not exist) but therefore VCV rack could not find the plugins created since it was not looking in this directory.

ricardomatias, 2023-09-11:
Same issue with mac, the plugin should be added to the folder named "plugins-mac-x64"

ohmtech-rdi, 2023-09-11:
Hi, thank you both for reporting.

Hmm I see a pattern here. Not sure how to handle that if I want to keep compatibility with previous versions of VCV Rack. I'll investigate on that. Please don't hesitate to comment if you find something relevant!

PR #629 (2023-12-07): Vcv rack sdk 2.4.0
This PR updates the `vcv-rack-sdk` submodule to version 2.4.0 since this version introduces a breaking change in the way plug-ins are loaded, and in particular their path. It also now supports the ARM64 architecture, and fixes a bug in the action system where the Windows binary was not installed.

- Fixes #619 
- Fixes #627 

## Todo

- [x] Check on macOS `x86_64`
- [x] Check on macOS `arm64`
closed 2023-12-07 by ohmtech-rdi