ohmtech-rdi/eurorack-blocks #627: Could not load plugin plugin.dylib (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))

andreij, 2023-11-08:
I'm running VCV Free Rack 2.4.1 https://github.com/VCVRack/Rack/blob/v2/CHANGELOG.md#241-2023-08-24 on a MacOs M2 (Sonoma).

I've followed the Ebb installation guide, and setup using:
```
erbb setup --with-xcode-support --with-vscode-support   
```

Then in `samples/drop` module:

```
$ erbb configure
$ erbb build
```

After that I run Simulation in VSCode, which copies the Plugin to the `Rack2/plugins` which seems apparently ignored by Rack startup, since the plugin name doesn't appear in log.txt. I quickly went through code and it's supposed to move plugins from that old directory to the new format `Rack2/plugins-XXX-YYY` but it doesn't seem so.

I've tried to manually move the `Drop` plugin to the new plugins directory and I get this error:

```
[0.611 warn src/plugin.cpp:196 loadPlugin] Could not load plugin /Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop: Failed to load library /Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop/plugin.dylib: dlopen(/Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop/plugin.dylib, 0x0006): tried: '/Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop/plugin.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop/plugin.dylib' (no such file), '/Users/andreamorandini/Documents/Rack2/plugins-mac-arm64/Drop/plugin.dylib' (mach-o 
```

I'm not an expert of compilers, so I've went through the Make process and the `x86` architecture is hardcode (I've tried to switch flags to ARM but ended in a rabbit hole) so it seems I'm missing some steps to how I'm supposed to configure Erbb for my machine?

I really love the framework you setup, so I hope to help to have this fixed, if possible.

ohmtech-rdi, 2023-11-13:
Hi @andreij 

Yes indeed, I need to support this. That's going to be a breaking change and not really happy about it. Didn't find time yet to work on it and to find a satisfactory solution, but I will.

Initially, as you noted, Rack didn't support `arm64` even though it was there already for quite some time.
When they wanted to add support to it, they went to change the path to the plug-in to include the architecture. They had an approach to use the library extension (dylib, dll, so) to imply the architecture and of course that was going to fail at some point.

In your case, a workaround in the meantime could be to run VCV Rack with Rosetta, and make sure the plug-in module is in the `x86_64` variant of the plug-ins folder.

andreij, 2023-11-13:
Thanks for your suggestion and thanks again for the wonderful platform you put in place!

PR #629 (2023-12-07): Vcv rack sdk 2.4.0
This PR updates the `vcv-rack-sdk` submodule to version 2.4.0 since this version introduces a breaking change in the way plug-ins are loaded, and in particular their path. It also now supports the ARM64 architecture, and fixes a bug in the action system where the Windows binary was not installed.

- Fixes #619 
- Fixes #627 

## Todo

- [x] Check on macOS `x86_64`
- [x] Check on macOS `arm64`
closed 2023-12-07 by ohmtech-rdi