ohmtech-rdi/eurorack-blocks #628: OSError: no library called "cairo-2" was found / no library called "cairo" was found

prettyboylaurel, 2023-11-14:
I'm following the "C++ on Windows with Visual Studio Code" guide on the eurorack-blocks website and everything seems to go smoothly until I run `erbb setup --with-vscode-support`, at which point I get this error.
```
$ erbb setup --with-vscode-support
Extracting msys2_mingw64.zip...
Extracting kicad_minimal_windows_6.0.11-0.tar.gz...
Extracting gcc-arm-none-eabi-10.3-2021.10-win32.zip...
Adding vscode support...
Installing python packages...
Optimizing erbb...
Traceback (most recent call last):
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\scripts\erbb", line 786, in <module>
    sys.exit (main ())
              ^^^^^^^
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\scripts\erbb", line 747, in main
    setup (args)
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\scripts\erbb", line 201, in setup
    setup.optimize_kicad_pcb_sch_read ()
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\setup\__init__.py", line 593, in optimize_kicad_pcb_sch_read
    from erbui.generators.kicad import pcb, sch
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\erbui\__init__.py", line 15, in <module>
    from .generators.vcvrack.panel import Panel as vcvrackPanel
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\erbui\generators\vcvrack\panel.py", line 15, in <module>
    from ..detail.panel import Panel as detailPanel
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\erbui\generators\detail\panel.py", line 39, in <module>
    import cairocffi
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\toolchain\python3-packages\cairocffi\__init__.py", line 48, in <module>
    cairo = dlopen(
            ^^^^^^^
  File "C:\Users\laurel\Desktop\eurorack-blocks\build-system\toolchain\python3-packages\cairocffi\__init__.py", line 45, in dlopen
    raise OSError(error_message)  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: no library called "cairo-2" was found
no library called "cairo" was found
cannot load library 'C:\Users\laurel\Desktop\eurorack-blocks\build-system\toolchain\msys2_mingw64\bin\libcairo-2.dll': error 0x7f
cannot load library 'libcairo.so.2': error 0x7e
cannot load library 'libcairo.2.dylib': error 0x7e
cannot load library 'libcairo-2.dll': error 0x7f
```
Things I have tried:

Uninstalling and reinstalling Python
Adding \eurorack-blocks, \Python and \Python\Scripts to the Path variable in Windows
Installing various versions of cairo that I found online and adding their locations to the Path variable

I am something of a novice with this kind of command line work so let me know if there's anything more I can add to help. Thanks!

ohmtech-rdi, 2023-11-15:
Hi @prettyboylaurel 

Thanks for your report.
We actually install our own version of cairo because this library is incredibly difficult to make work right. So your problem is half a surprise unfortunately.

Here it finds the library, but cannot load it, as shown in:
```
cannot load library 'C:\Users\laurel\Desktop\eurorack-blocks\build-system\toolchain\msys2_mingw64\bin\libcairo-2.dll': error 0x7f
```

So I would suspect the dll is not compatible with your machine.

Could you please tell me:
- Your version of Windows,
- The actual architecture of your computer running Windows? I suspect this it not an Intel chip, but maybe an ARM?

Thanks!

prettyboylaurel, 2023-11-15:
Thanks for the response! I'm running Windows 10 Version 22H2 (OS Build 19045.3570), and you're right that it's not an Intel CPU, but it isn't ARM, it's an AMD Ryzen 5 2600X. Let me know if you need any more information.