ohmtech-rdi/eurorack-blocks #673: VCV Rack crashes when I use qu.quantize Faust library

bepitulaz, 2024-03-28:
Here's my Faust program
```
import("stdfaust.lib");

// VCO 1
vco1_gain = vslider("vco1_gain", 0, 0, 1, 0.01);
vco1(freq) = os.m_osccos(freq) * vco1_gain;

// VCO 2
vco2_gain = vslider("vco2_gain", 0, 0, 1, 0.01);
vco2(freq) = os.phasor(0.2, freq) * vco2_gain;

// Control voltage input
cv = hslider("pitch", 0, -5, 5, 0.01);

// Convert CV to frequency (1V/Octave)
cv_to_freq(cv) = 440.0 * pow(2, cv); // 440Hz is the frequency for A4

process = cv : cv_to_freq : qu.quantize(440, qu.penta) <: (vco1 + vco2);
```

And, this is the erbui file:
```
control vco1_gain Pot {
      position 3hp, 34mm
      style rogan, 2ps
      label "GAIN"
}

control vco2_gain Pot {
      position 9hp, 34mm
      style rogan, 2ps
      label "GAIN"
}

control pitch CvIn {
      position 3hp, 111mm
      style thonk.pj398sm.knurled
      label "1V/OCT"
}

control audio_out AudioOut {
      position 9hp, 111mm
      style thonk.pj398sm.knurled
      label "OUT"
}
```
When I load the module, then VCV will crash. I tried removing the quantizer like this below, then it will work. The problem is I need to use the quantizer library.
```
process = cv : cv_to_freq <: (vco1 + vco2);
```

Here's the full report when it crashes hope it helps. I'm not good to debug C++ code, maybe you can guide me how to debug it to get better hint about the issue. Thanks.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Rack [98192]
Path:                  /Applications/VCV Rack 2 Free.app/Contents/MacOS/Rack
Identifier:            com.vcvrack.rack
Version:               2.4.1 (2.4.1)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-03-28 22:52:14.4639 +0200
OS Version:            macOS 14.3.1 (23D60)
Report Version:        12
[... 455 lines cut ...]
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "a3f9ec09-b145-49f3-8142-da4a1f338456",
      "experimentId" : "65f21aa774b6f500a45abd7d",
      "deploymentId" : 400000002
    }
  ]
}
}

Model: Mac14,7, BootROM 10151.81.1, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M2, Apple M2, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Aug 26 2023 18:16:13 version 18.20.439.0.7.8.163 FWID 01-49626b4b
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

bepitulaz, 2024-03-29:
Okay. I managed to learn about attaching the debugger to VS Code. And, it throws this error. A progress, I got a meaningful error.
```
Warning: dsp_memory_manager::destroy: Potential memory leak.
         Set a breakpoint to debug.
Rack(3277,0x1e2a7dc40) malloc: *** error for object 0x120800150: pointer being freed was not allocated
Rack(3277,0x1e2a7dc40) malloc: *** set a breakpoint in malloc_error_break to debug
```

bepitulaz, 2024-03-30:
Another update for this issue. If I generate the same code from Faust directly to VCV Rack without `erbb`, then it will work perfectly.

ohmtech-rdi, 2024-05-06:
Hi @bepitulaz 

Sorry for the late reply, we are quite busy here.

Did you try to set a breakpoint on `malloc_error_break`? This should lead to where the error occurs, at least to know if it's somewhere into Faust code or actually unrelated.