ohmtech-rdi/eurorack-blocks #696: Any way to also prototype display code?

Ericxgao, 2025-04-11:
Hi folks,

Awesome work on this framework. I will probably be making an attempt to make my first module with this soon! I was curious if there was a way to prototype the display code as well, I often use u8g2 in other hardware libraries in the past, would be awesome to be able to also render that in VCV Rack to test.

Thanks!

ohmtech-rdi, 2025-05-14:
Hi Eric,

You do have some support for it already, for example the [`Display`](https://github.com/ohmtech-rdi/eurorack-blocks/blob/main/include/erb/Display.h) class and [Eastrising 128x64 display](https://github.com/ohmtech-rdi/eurorack-blocks/tree/main/build-system/erbui/generators/front_pcb/er.display.spi.128x64.manual)

Though for this, since `kivu12` doesn't have display support, you need to make a custom board and route the SPI to the display.

We use this for our products to prototype the display in the simulator, but I don't have enough time now to make sample code, and it is quite involved.

I'll try to do this at some point. Thanks also for the nice words!

Ericxgao, 2025-06-05:
Thanks! Curious if there's any recommended way to set up in eurorack blocks and add display code after? I can work with no display in VCV Rack for now.