ohmtech-rdi/eurorack-blocks #697: TypeError in panel generator: SVG opened in text-mode under Python 3

liamge, 2025-05-14:
When running the VCV Rack simulator debug under Python 3, the panel generator opens SVG files in text mode. CairoSVG’s gzip-header check then does:

`if bytestring.startswith(b'\x1f\x8b'):`

but since `bytestring` is a Unicode str, you get:

`TypeError: startswith first arg must be str or a tuple of str, not bytes`

# Environment

macOS: Sonoma 14.5
Python: 3.13.3

# Proposed Fix:
In `build-system/erbui/generators/detail/panel.py`, change `generate_image(...)`:

```
def generate_image(self, context, module, render_mode, image):
        # …mode checks…
-       with open(image.file) as file:
-           tree = cairosvg.parser.Tree(file_obj=file)
+       # Open SVG in binary mode so CairoSVG sees bytes, not str
+       with open(image.file, 'rb') as file:
+           tree = cairosvg.parser.Tree(file_obj=file)
```

After applying the patch, you may need to delete any stale __pycache__ in build-system/erbui so Python picks up the change.

Thanks for maintaining this project—I’m happy to submit a PR if that helps!

ohmtech-rdi, 2025-05-14:
Hi Liam,

Thanks for the detailed report and fix suggestion, and you are welcome to make a PR.

I'm not sure why Python wouldn't pick up the change? Can you please elaborate on that?

Thanks!

ohmtech-rdi, 2025-05-14:
So that seems to have been introduced 2 days ago with 2.8.0. I'm not sure why this fails only now, because the test to check for gzip compression is here since 2 years apparently.

This is actually a blocker for my company, so I'll make the PR now.

closed 2025-05-14 by ohmtech-rdi

liamge, 2025-05-14:
No worries! Sorry the pycache stuff was in my internal notes to remind myself when I tried the spotcheck fix, disregard!

PR #698 (2025-05-14): Read as binary, as the gzip check in cairo is expecting bytes, not str
This PR fixes a problem of Eurorack-blocks using Cairo:
```
TypeError: startswith first arg must be str or a tuple of str, not bytes
```

The test [used to be in 2.7.1](https://github.com/Kozea/CairoSVG/blob/2.7.1/cairosvg/parser.py#L388):
```
if len(bytestring) >= 2 and bytestring[:2] == b'\x1f\x8b':
```
But is [now with 2.8.0](https://github.com/Kozea/CairoSVG/blob/2.8.0/cairosvg/parser.py#L388C13-L388C51) (released 2 days ago):
```
if bytestring.startswith(b'\x1f\x8b'):
```
Which triggers the error.

Cairo was always exp [... cut]